LBCC-SpaceClub/HAB2017 #1: Updates for the manual motor adjustments area

lo9key, 2017-05-17:
Changing the range on the manual motor adjustments from 0-100 to 0-360 would let us easily visualize those values as degrees.  That would be great for manually moving the dish to specific angles during testing.

The text box is great, and would be even better if we could type values in there in addition to using the sliders.  That would be useful if we want a really specific angle like 36.71 degrees.

It might also be a good idea to label both sliders as azimuth and elevation.

[image]

proutyio, 2017-05-17:
Thanks for this input this was exactly what I was gonna ask if we needed.
 I will make these changes

proutyio, 2017-05-18:
this is implemented, just need to do some input verification still

closed 2017-05-18 by proutyio

proutyio, 2017-05-18:
Input validation working. Issue closed.