LBCC-SpaceClub/HAB2017 #2: Add thread (?) for motor outputs

lo9key, 2017-05-17:
The GUI is really coming together and it's probably time to add outputs for the motors.  I'm using this issue thread to outline my thoughts and general plan.

Ideally this software will be able to control trackers using either stepper or servo motors.  The hardware for each motor type is very different and we're still sorting out several important details:

- Stepper motors will probably be controlled by an arduino connected via ethernet.  The IMU and GPS will probably be on the same arduino as the stepper outputs. This is necessary due to the ship station being 100' away from the GUI laptop.
- Servo motors are currently controlled by serial output.  IMU and GPS inputs are on a different arduino connected via USB. The GUI laptop can be within a few feet of the ground station.  Ideally this remains compatible with the standard MSU trackers.

I see that @proutyio already added radio buttons for Station GPS Data to toggle between Arduino USB and Arduino Ethernet.  That looks like a good way to toggle between both local GPS positions and how we output to the motors.

So now the question is.. do those buttons toggle between different threads, or just set a flag that toggles between algorithms?

The servo algorithm is roughly to, several times per second:

1. Read new local IMU and GPS information from serial.
2. Calculate pointing solution as azimuth and elevation angles.
3. Convert from degrees (0-360) to servo number (0-255).
4. Output servo numbers to each servo.

The stepper algorithm is roughly:

1. Read any data over ethernet, update GUI elements with local GPS, IMU, and solution azimuth and elevation angles.
2. Every time we get new (different) payload GPS information:
- Send payload GPS over ethernet.

lo9key, 2017-05-17:
Thinking about it, I guess both algorithms could run at the same frequency of about once per second.  If the payload GPS information hasn't changed, the servo algorithm could just stop right there.  It only needs to write to the servos when we get new information, because the ground station is designed to be on stable ground.  The servo station doesn't need to handle the motion of the waves like the stepper station does, so it doesn't need a fast loop.

lo9key, 2017-05-22:
I made a new branch for this change so that Master can still be used in the meantime.  @proutyio feel free to edit my branch if you agree with the direction it's going.

Right now I have it set up to call the ArduinoThread class differently based on which button is used.  The interval is checking for that manual local gps values button and calling the ArduinoThread.pointTo function using the correct values.

The ArduinoThread's init function determines if it should use servos or steppers, and uses that to create either a ServoControl or StepperControl object.  Almost all of the code in ArduinoThread should be able to remain the same for both.  We'll see.

If you think I'm going down the wrong path here that's ok too, just make your own branch if you want to explore another approach.  We can wait to merge this feature to master until we're both happy with it.

proutyio, 2017-05-23:
I was looking at the changes you are making in your branch and here are some of my thoughts.
# Connect to the arduino and start a thread to keep this class updated
 +		self.arduino = self.connectToArduino()
 +		if self.arduino:
 +			Thread.__init__(self)

This code needs to happen in a thread. Thread.__init__ is not starting the thread. Once .start() is called the thread will spawn and execute the code in the run() method. The thread only exsists as long as the run() method is still alive. 
If we dont connect to the arduino in a different thread then the gui is going to be waiting for it to connect. The run method should start the connection to the arduino just before it goes into the while loop and if it fails then the run method ends which ends the thread. Then we could test for a failure and pop the object off the stack to start the connection over.

lo9key, 2017-05-23:
As far as I know, in [AntennaTracker.startArduino()](https://github.com/LBCC-SpaceClub/HAB2017/blob/aefc0757b798ba7e1585c1b0ba39637c0ae30e55/AntennaTracker/AntennaTracker.py#L96-L121), I'm calling the start() on whichever object was selected by the gui.

So toggling the radio button causes either a StepperControl or ServoControl object to be added to the arduino_list.  The same function then calls start() on whatever is in there.

As long as both StepperControl and ServoControl implement the same public functions, AntennaTracker can use either one without worrying about the details.  My goal here was to abstract away the hardware implementation, all the gui needs to know is that self.arduino_list[0].getLat() returns the station's latitude.

Oh, ignore ArduinoThread for now.  I renamed it ServoControl but haven't cleaned it out yet.  This is still a very experimental branch.

lo9key, 2017-05-25:
@proutyio I have it in a pretty good place right now.

- [X] Reorganized the threads so that each child (arduino/database) can update the GUI directly, which makes it a lot more responsive because they don't need to wait for the maintenance loop to update values.
- [X] Connects to the servo arduino, reads gps and IMU values and updates GUI text boxes.
- [X] Connects to the servo maestro board, can move servos.
- [X] In ServoControl there are Arduino and Servo classes, but they are only used in ServoControl.
- [ ] ServoControl and StepperControl should share any variables and function names that might be used by AntennaTracker. I'm thinking things like pointTo(), isConnected(), things like that.
- [ ] Calculate the target angles and update the GUI elements for the compasses.  This has to be done in ServoControl and StepperControl classes because the algorithm is different for each.
- [ ] Do more testing to make sure it handles edge cases like not finding the usb ports, or what happens when things are unplugged while being used.  I've been doing this as I go but there are a lot of gotchas here.
- [ ] Repeat most of these steps for StepperControl

proutyio, 2017-05-25:
Nice work dude! That all looks great

lo9key, 2017-05-25:
Just added the calculations and both GUI elements update now:
[image]

@proutyio are you happy with this feature branch? I know it's probably hard for you to test anything about it until we get you an arduino+gps+imu to play with, but I'd like to merge it into master soon.  With any luck we might be able to try using this on Saturday.

Will you be able to make it this weekend? I'll give you the USB arduino I've been testing ServoControl with, and I'll switch over to the ethernet version to start working with StepperControl.

lo9key, 2017-05-25:
Servo outputs are implemented.  We'll need to verify the min/max range for both axis, in the meantime I used the defaults from MSU.  Hopefully that's not mistake!

proutyio, 2017-05-26:
Good work man, you should merge anything you have. I plan on looking through stuff this afternoon and doing some more work on it.

lo9key, 2017-05-26:
I went ahead and merged my branch to master.  Closing this issue for now.  We can create new ones to deal with specifci bugs in the implementation.

closed 2017-05-26 by lo9key