LBCC-SpaceClub/HAB2017 #3: Odd import issue with python3

lo9key, 2017-05-21:
Going back to my desktop, I found a hiccup/issue with imports.  I'm using this issue to report and learn about it.

Apparently, pip installs packages for python 2.x and we need to use pip3 to install packages for python 3.x.

That's not a deal breaker, but pip3 has to be installed separately, and it choked on my machine:

```
$ make
pip3 install -r requirements.txt
Collecting PyMySQL==0.7.11 (from -r requirements.txt (line 1))
  Downloading PyMySQL-0.7.11-py2.py3-none-any.whl (78kB)
    100% |████████████████████████████████| 81kB 1.1MB/s 
Collecting Kivy==1.9.1 (from -r requirements.txt (line 2))
  Using cached kivy-1.9.1.tar.gz
    Complete output from command python setup.py egg_info:
    Using distutils
    
    Cython is missing, its required for compiling kivy !
    
    
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-9erhxldg/Kivy/setup.py", line 184, in <module>
        from Cython.Distutils import build_ext
    ImportError: No module named 'Cython'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-9erhxldg/Kivy/
Makefile:4: recipe for target 'install' failed
make: *** [install] Error 1
```

After fiddling with it I was able to get everything to work without using pip3, except the pymysql package, which I had to install using `sudo apt-get install python3-pymysql`

Before we release this for others, we should nail down this installation process so that people can just run a command without needing to hunt down these libraries.  I'm a big fan of using makefiles for this sort of thing, but I'm open to other ideas.

lo9key, 2017-05-21:
I think we should probably just force people to use either python 2 or 3 and go from there.  Trying to support both will be a headache.

I haven't looked into the differences but I think you said kivy runs best on python 3?  Should we just stick with that?

lo9key, 2017-05-21:
See ed76ee2a70a5fb9949a2a33c8fb8e52359def423

proutyio, 2017-05-25:
I agree about forcing python3. No reason to support old libraries if we are looking at maintaining this software into the future.