LBCC-SpaceClub/HAB2017 #5: Update compass graphic to show only 0-360 moving clockwise from the top.

lo9key, 2017-05-21:
Right now the graphic has two ranges of degrees which might get confusing.

This is a small thing, but I don't want to forget it.

proutyio, 2017-05-23:
I updated it, let me know what you think

lo9key, 2017-05-23:
They look great, thank you.  Some of the numbers are a little small but what can you do, right.

If you're bored and have time (haha yeah right!), it might be worth seeing what it looks like to remove some of the numbers.  For example, keeping 0, 45, 90, 135, 180 etc but removing numbers like 10, 20, etc.

Totally not important though compared to other features.

proutyio, 2017-05-25:
Ya I think I will make one from scratch when I have time to get the values more hi res so they are easier to see, less ticks sounds good too.